spacetelescope/jwql #1222: NIRCam readnoise monitor not showing readnoise image and hasn't run lately

bhilbert4, 2023-03-20:
The image of difference between the latest readnoise and the readnoise reference file is not showing up in the app. 

The readnoise plots above it indicate that the only results we have are from last November. Is that correct? (Also, there are no results at all before last November, which seems incorrect.) 

It'd be worth checking the logs and checking if the output files exist in the correct directories. Maybe they aren't moving from pljwql2 to pljwql1? The same is true for all instruments except NIRISS.
@bsunnquist @york-stsci

york-stsci, 2023-03-20:
This is not (or should not be) an issue with where things are running. Except during specific test and debugging scenarios, everything is still running on pljwql1. I would like to be *able* to change that at some point in the future (especially if we discover that pljwql1 is becoming too loaded down between running monitors and serving requests), but certainly we haven't made that change over up until now.

The regression test monitor has been running through without crashing since late 2022, and as far as I can tell if it's found new data it's run that data and then stored it.

I suspect that part of the issue may be that, before the celery task server was in place, the monitor was always eventually crashing, and so it might end up storing in the history table that it had covered a particular exposure, but not been able to actually store results for that exposure. That said, I don't know that's the issue, just that the timing is suspicious.

york-stsci, 2023-03-20:
Looking at the March 20 log, NIRISS was the only instrument that had exposures that the readnoise monitor actually found, and that weren't already in its database.

york-stsci, 2023-03-20:
What's been happening since November, as far as I can tell, is that the readnoise monitor does the following:

- looks at the history database to see when its most recent query was run
- Retrieves any more recent files that match some criteria (I don't know, off-hand, what they are)
- For any retrieved files, calibrates them and stores them into its stats table

Right now, for every instrument except NIRISS, it's either not finding any new files that match its criteria at all, or any files that it finds are already in the stats database, and so aren't added. So this could be an error with its matching criteria, it could be an error with figuring out if a file is already in its database, or it could be that the way it retrieves display information isn't working.

One possible thing is that:

- It doesn't check for files that are older than its newest history entry
- There are files that it never stored in the stats database (because of some error) that it's now just not looking at
- When the webpage script displays files, it doesn't display files in the stats database that are more recent than the last 
  entry in the history database, even if they are in the stats database
- For some reason there are files that are in the stats database that don't cause it to update its history database, and 
  because of that they're never going to be displayed on the webpage.

And, of course, there are many other options too.

york-stsci, 2023-03-20:
For what it's worth, my main thought about why NIRISS is working is that, before we had the celery server in place, I'm guessing the monitor never got as far as NIRISS before crashing, so when it first got there with the celery server in place, it was more or less starting fresh, and didn't have any inherited inconsistencies. If that's right, then the ultimate solution for the other instruments might be to reset their databases and re-populate them one at a time.

bhilbert4, 2024-05-01:
Fixed

closed 2024-05-01 by bhilbert4