spacetelescope/jwql #1232: Look into pytest-django

bhilbert4, 2023-03-30:
Look into using pytest-django to allow us to run tests involving the django models. We currently skip these tests on github actions since they have no access to the models.

bhilbert4, 2023-06-02:
This seems like something we don't need to worry about if we're going to move from Django models to postgres, right? @BradleySappington are we going to keep any Django models after the transition?

BradleySappington, 2023-06-02:
This is still relevant as we are still going to continue using Django Models.  The models are currently created (for us by django) as tables in sqlite.  In order to get them to fulfill our needs we will need to update the Django Models to be stored in tables using a postgres db.

bhilbert4, 2023-06-02:
Ah I see. I had been assuming we were going to move everything to postgres tables and access similarly to how we do the monitor database tables. Thanks for setting me straight!

bhilbert4, 2024-05-01:
@BradleySappington is this now even more relevant than before since we have everything in Django models now?

BradleySappington, 2024-05-02:
yup!  We need to make sure we are testing all of these models