spacetelescope/jwql #1251: Remove MAST monitor

bhilbert4, 2023-04-17:
As far as I can tell, the mast monitor runs each day, but the results are not displayed anywhere. I just looked at the html files generated by today's run on the ops server. The monitor creates a couple of bar charts showing the number of files present in MAST for each file type (image, spectrum, cube).  The plots look like they are incorrect and missing a large amount of data.

Given the very similar tracking done by the filesystem monitor, maybe we should just turn off and remove the mast_monitor? 
@mfixstsci what do you think?

mfixstsci, 2023-04-17:
@bhilbert4 I am not opposed to this idea. No need to duplicate the figures or the work on the servers when those resources could be used somewhere else. How much work would it be to disable the monitor? Besides the mentioned figures, are there other features that the instrument teams use regularly we would want to keep?

bhilbert4, 2023-06-02:
Resolved via #1256

closed 2023-06-02 by bhilbert4