spacetelescope/jwql #1269: Construct a database containing info from headers

bhilbert4, 2023-05-25:
From NIRSpec:

It could be useful to have a database that contains header keyword information from all files. This would make it relatively quick and easy to plot metadata as part of, or independent from, instrument monitors.

@melanieclarke

bhilbert4, 2023-05-25:
I agree that this could be a very useful addition to JWQL. On thing to consider first: Is it possible to get this information through any other means? MAST allows you to create queries using JWST instrument keywords, which translate to many of the header keywords. On the portal, these queries return the names of exposure files that match the values of the input keywords. But is it possible to use MAST to say, plot the value of a particular keyword in all files over the last 6 months? It's not clear to me if this can be done or not. If not, that's an argument in favor of having a JWQL header keyword database.

melanieclarke, 2023-05-25:
Thanks @bhilbert4 

To be clear - the request is to store the headers themselves in their entirety, as text blobs, rather than a pre-determined set of known keyword-value pairs. The idea is that that arbitrary information from them can then be retrieved on the fly, without requiring database table changes or specific accommodation in the code.

mfixstsci, 2023-11-08:
Hey @melanieclarke! @BradleySappington and myself are looking at this issue (Brad will be working on it) but I am curious what you mean by **ALL** files in your comment. Are we talking about every file available for NIRSpec from uncal to level 3 products, which header extensions (0th, 1st, all of them, data too?)

BradleySappington, 2023-11-22:
Spoke with @sosey, Discussion should occur with MAST folks about current functionality, or future plans for them to accommodate this request on their end.

sosey, 2023-11-22:
@melanieclarke let's chat next week in a little more detail about the nirspec team needs.