spacetelescope/jwql #1275: Automated check for failed jobs

bhilbert4, 2023-05-30:
We've seen cases where one of the JWQL cron jobs was interrupted or failed in some way, and the corresponding lock file was not removed. This means that subsequent runs of the job won't be able to start. Currently when this happens, the developers are not alerted in any way, and only find out when someone goes looking for the outputs from the job.

It would be nice to implement an automated check for interrupted/failed jobs, and alert the developers when it happens. 

For archive_database_update.py and generate_preview_images.py (i.e. jobs that do not use a celery/redis lock, but instead a manually constructed lock file), we could set up a job to look at any lock files in the directory and check their age. Any lock files older than some threshold may indicate a problem and should trigger a notification to the developers.

What would be a good threshold to use for this? It seems like even just 24 hours would be a conservative estimate. For normal operations, archive_database_update and generate_preview_images both complete in significantly less time than 24 hours. And in special cases where those tasks are catching up on a backlog of data and do take 24 hours, the developers would already know about the situation and could ignore the notification.

What about tasks that use the redis lock? I'm assuming there is some way to check the age of a given lock...
@york-stsci 

@melanieclarke @mfixstsci

york-stsci, 2023-05-30:
For tasks that use redis locking I’m not sure that it’s possible to find lock age, but it is possible to set a lock to expire after a set interval, which would probably be as good for most purposes.

Brian

mfixstsci, 2023-05-30:
@york-stsci I think that would work for what we need. We are thinking 2 days should be enough time assuming that when its running and caught up it should only take a couple of hours to run.

closed 2023-06-05 by mfixstsci