spacetelescope/jwql #1285: TEL tools for multi-instrument effects? PSF metrics, etc.

bhilbert4, 2023-06-28:
Strehl ratio, encircled energy, etc.

The TEL branch may come to us with a request to incorporate some of their tools into JWQL so that they can be automated.

Matt Lallo is the contact for this.

bhilbert4, 2023-09-29:
@Skyhawk172 I missed your talk this week, but others on JWQL saw it and wondered if tools for tracking PSF metrics would be something you'd want to implement in JWQL.

tcomeau, 2023-09-29:
Matt's a contact, I'm also a contact (Tom Comeau) and we do have some things we'd like to try.

Skyhawk172, 2023-10-05:
Essentially, as a first pass, we'd like to do something like what we did during commissioning, in our OTE-28 "in-focus PSF checks" activity: For various instruments/detectors, we'd like to measure metrics such as encircled energy, FWHM, etc and possibly generate empirical PSFs so that we can monitor any trends in the image quality over the field of view. The SI teams are probably already doing something like this, so we might be able (with your help) to re-use some of what's already being done. Otherwise, we have notebooks to calculate some metrics of interest.

One hurdle for us has been getting and storing the data for _all_ the SI/detectors, so hopefully JWQL can do that for us easily. 

In any case, that's a high-level summary for now. Let us know your thoughts on our plan and whether any of it is already implemented by the SI teams.