spacetelescope/jwql #1286: Allow for more fine-tuned file searching

bhilbert4, 2023-06-28:
From @aliciacanipe:

When using JWQL as a first check for help-desk tickets for data issues, it would be helpful to have more fine-grained file searching abilities. e.g. adding visit, subarray, filename other useful headers as search options

This seems like it would be dependent upon #1269

BradleySappington, 2023-11-27:
@bhilbert4 - its not looking like we are going to keep header text blobs in the DB.  Is there specific header data you want stored in with the rootfile info?

bhilbert4, 2023-11-27:
visit would be nice. Although I'm not sure how this would work in practice on the query page. We probably don't want to list all possible visit number values across all programs? And if we were  to list visit numbers, we'd probably want to have observation numbers as well. Maybe we could adjust the "FIND A JWST PROPOSAL OR FILE" box on the front page to pay attention to optional observation/visit numbers? And/or put a copy of that on the query page so that it can be combined with e.g. a filter to search for?  @aliciacanipe, any other properties you'd want to be able to search by? 

It also looks like some nircam subarrays are missing from the current list. e.g the TA subarrays.

BradleySappington, 2023-12-15:
#1412

closed 2024-01-04 by BradleySappington