spacetelescope/jwql #1303: Move date range archive to the query page

melanieclarke, 2023-08-02:
Currently, date range queries are handled by the date range archive page (archive_date_range).  This allows searching by date and some sorting/filtering of the results afterward, but it does not allow combinations of date range with other filters to narrow the search results.

Now that the query page is working well, it would be nice to retire the archive_date_range page and move date filtering to the general query page.  This would be an added benefit to general queries, which currently do not allow date filters, and to date range queries, which currently do not allow general filters.

bhilbert4, 2023-10-06:
This is high priority.
(Test message to see if the jira-github bot picks up this issue)

stscijgbot-jwql, 2023-11-15:
_Comment by [Bradley Sappington](https://jira.stsci.edu/secure/ViewProfile.jspa?name=bsappington) on [JIRA](https://jira.stsci.edu/browse/JWQL-142?focusedCommentId=670859#comment-670859):_

#1396

Ready for testing, currently block with MAST build issue

stscijgbot-jwql, 2023-11-15:
_Comment by [Bradley Sappington](https://jira.stsci.edu/secure/ViewProfile.jspa?name=bsappington) on [JIRA](https://jira.stsci.edu/browse/JWQL-142?focusedCommentId=671514#comment-671514):_

[Bryan Hilbert](https://jira.stsci.edu/secure/ViewProfile.jspa?name=hilbert) This is in Develop branch, can be merged to test after our release is finished

closed 2024-01-12 by stscijgbot-jwql