spacetelescope/jwql #1314: Use Ruff CI

stscijgbot-jwql, 2023-09-05:
_Issue [JWQL-55](https://jira.stsci.edu/browse/JWQL-55) was created on JIRA by [Bryan Hilbert](https://jira.stsci.edu/secure/ViewProfile.jspa?name=hilbert):_

Switch to using Ruff as the linter for new PRs. Previously we used pep8speaks as our linter, but this stopped working. From our research Ruff looks like the preferred way to go.

bhilbert4, 2024-05-10:
We've moved away from this now that pep8speaks is working again.

closed 2024-05-10 by bhilbert4

stscijgbot-jwql, 2024-05-13:
_Comment by [Bryan Hilbert](https://jira.stsci.edu/secure/ViewProfile.jspa?name=hilbert) on [JIRA](https://jira.stsci.edu/browse/JWQL-55?focusedCommentId=698862#comment-698862):_

This work was put on hold when pep8speaks began working again. For now, we're happy with pep8speaks.