spacetelescope/jwql #1354: Intermittent pipeline failures

stscijgbot-jwql, 2023-09-08:
_Issue [JWQL-136](https://jira.stsci.edu/browse/JWQL-136) was created on JIRA by [Bryan Hilbert](https://jira.stsci.edu/secure/ViewProfile.jspa?name=hilbert):_

When looking at bias monitor log files, I'm seeing some intermittent pipeline failures that I don't understand. For example, on Sept 3, 4, and 6, 2023, the log files report that the pipeline failed when trying to process jw01453019001_03301_00001_nrca1. But the log file from Sept 7 shows that the pipeline run was successful, and the results added to the bias monitor database table.

It's not clear what's going on here. Could it be a problem copying files from one directory to another, similar to the problem seen with some of the dark files during dark monitor runs? To help diagnose that issue, PR [#&#x2060;1301](https://github.com/spacetelescope/jwql/pull/1301/) added some lines that check the file size of the original vs the copied versions of a file and throws the file out if it was not fully copied. Maybe we need to add something similar to the bias monitor? Or perhaps somewhere in shared tasks, so that it can be applied to all monitors?

york-stsci, 2023-09-11:
Just starting to look at this, but I do know that FITS files have built-in checking (that you'll sometimes see in logs). The FITS header contains some information on the expected size of the FITS file, so if you use `astropy.io.fits` to open a FITS file, you will sometimes see an exception like "file size was <X> but expected size was <Y>", and if we're not seeing errors of that sort then that's probably not the problem.

bhilbert4, 2023-09-11:
For #1301 that does seem to be the case. Comparing the file sizes of the files in the transfer directories to those in MAST, where they were copied from, showed that the copies were incomplete. In that case, the errors in the logs (which you can see an example of in #1301) looked to be basically stdatamodels trying to interpret the error from astropy.io.fits when the latter tried to retrieve the data.

bhilbert4, 2023-09-11:
These calls were coming from the bias monitor.

york-stsci, 2023-09-11:
And at least one other place, because the bias monitor was sending in _uncal_0thgroup.fits, but there are also calls to calibrate _uncal.fits, which would not be coming from the bias monitor, but which seem to fail in the same way.

Honestly, I think my first step here is to add a bunch more logging to the script that actually *runs* the pipeline, and see what that tells us. The big issue is that, given the failures are the result of all of the calibration steps being skipped, *why* is it deciding it has to skip all of the steps, and then why does it decide later to do them. There is, very clearly, something strange going on, but figuring out what it is will be a task.

york-stsci, 2023-09-12:
Okay, here's my interim report:

- Looking at the celery task logs, the new output code is working, and we're getting the additional output content.
- We have not had any failed calibrations yet, so I can't currently see what caused them

As such, this issue is probably going to be hanging out in waiting mode for the next little while, until we actually have a failed pipeline run to look at.

stscijgbot-jwql, 2025-01-28:
_Comment by [Bryan Hilbert](https://jira.stsci.edu/secure/ViewProfile.jspa?name=hilbert) on [JIRA](https://jira.stsci.edu/browse/JWQL-136?focusedCommentId=941845#comment-941845):_

Closing this since it has been a while and we haven't seen any problems. If we end up seeing more pipeline failures in the future, we can open a new issue.