spacetelescope/jwql #1357: Dark monitor crashing due to reported missing env packages

stscijgbot-jwql, 2023-09-13:
_Issue [JWQL-138](https://jira.stsci.edu/browse/JWQL-138) was created on JIRA by [Bryan Hilbert](https://jira.stsci.edu/secure/ViewProfile.jspa?name=hilbert):_

On Sept 13, the dark monitor crashed while trying to create png files. The error statement reports:

"RuntimeError: Neither firefox and geckodriver nor a variant of chromium browser and chromedriver are available on system PATH. You can install the former with 'conda install -c conda-forge firefox geckodriver'."

However, I've confirmed that both firefox and geckodriver are present in the jwql-3.9 environment. The environment was just deleted and re-created from scratch last week. Could it be that the web service is working with an old version of the environment? Except that both packages were present in the old version of the environment as well.

bhilbert4, 2023-09-21:
I've found a way to get this working, although I still don't understand why what we had been using no longer works. The packages are in the environment, and the correct path to the packages is in the PATH env variable. 

Anyway, when I supply an instance of a web driver in the call to export_png, it works. This suggests that for some reason it isn't able to find or create a web driver instance otherwise?

This works, as tested on the ops server:
```
from bokeh.plotting import figure
from bokeh.io import export_png
import numpy as np
from selenium import webdriver
from bokeh.models import LinearColorMapper
plot = figure(title='test',tools='')
image = np.zeros((200,200))
image[100:105, 100:105] = 1
ny, nx = image.shape
mapper = LinearColorMapper(palette='Viridis256', low=0 ,high=1.1)
imgplot = plot.image(image=[image], x=0, y=0, dw=nx, dh=ny, color_mapper=mapper, level="image")
options = webdriver.FirefoxOptions()
options.add_argument('-headless')
driver = webdriver.Firefox(options=options)
export_png(plot, filename='test.png', webdriver=driver)
driver.quit()
```

bhilbert4, 2023-09-21:
Further digging has revealed that this issue is coming from updated versions of selenium. With selenium at 4.9.1, everything works fine (without having to supply a webdriver instance). Using 4.10.0, bokeh's export_png fails. Just before that failure is a statement `The version of chrome cannot be detected. Trying with latest driver version`. And then in versions after 4.10.0, export_png fails without that statement. 

So it seems that in order to keep up with the latest version of selenium, we'll need to use the updated code above when creating png files.

closed 2023-10-02 by mfixstsci

stscijgbot-jwql, 2024-05-06:
_Comment by [Bryan Hilbert](https://jira.stsci.edu/secure/ViewProfile.jspa?name=hilbert) on [JIRA](https://jira.stsci.edu/browse/JWQL-138?focusedCommentId=697824#comment-697824):_

Resolved via [#&#x2060;1360](https://github.com/spacetelescope/jwql/pull/1360)