spacetelescope/jwql #1395: Data created by monitors and accessed by the JWQL web server should be on central storage

york-stsci, 2023-11-07:
See JIRA JWQL-153 for a full description (https://jira.stsci.edu/projects/JWQL/issues/JWQL-153)

york-stsci, 2023-11-07:
As planned, this change will involve several steps:

1. Add a "working" key to config.json, with an identical value to the current "output" key
2. In all JWQL code that uses the "output" key,
    - If the code is finding a file to display on the website, leave it as is
    - If the code is storing a file that the website will display, leave it as is
    - If the code is storing a file that is not needed by the website, change the code to use the "working" key
3. Copy the files that are used by the web server to central storage, based at /grp/jwst/ins/jwql/data rather than at /internal/data1/data
4. Set the "output" key to point to central storage
5. Restart the webserver

In terms of timing, steps 1 and 2 can be done on their own, and should be merged before any later steps run. Once the code changes are in place, steps 3 and 4 should be run while none of the monitors are running, probably during the day. Steps 3 to 5 should be run, essentially, as a unit, to make sure that all outputs are in place, and the directories are synchronized when the changeover happens.

stscijgbot-jwql, 2024-05-09:
_Comment by [Bryan Hilbert](https://jira.stsci.edu/secure/ViewProfile.jspa?name=hilbert) on [JIRA](https://jira.stsci.edu/browse/JWQL-153?focusedCommentId=698439#comment-698439):_

[Brian York](https://jira.stsci.edu/secure/ViewProfile.jspa?name=york) is this still in progress, or has this all been completed?

stscijgbot-jwql, 2024-05-10:
_Comment by [Brian York](https://jira.stsci.edu/secure/ViewProfile.jspa?name=york) on [JIRA](https://jira.stsci.edu/browse/JWQL-153?focusedCommentId=698595#comment-698595):_

[Bryan Hilbert](https://jira.stsci.edu/secure/ViewProfile.jspa?name=hilbert) yes this has been done.

closed 2024-05-10 by stscijgbot-jwql