spacetelescope/jwql #1415: Dark monitor plots do not respect readout pattern

stscijgbot-jwql, 2023-12-20:
_Issue [JWQL-157](https://jira.stsci.edu/browse/JWQL-157) was created on JIRA by [Bryan Hilbert](https://jira.stsci.edu/secure/ViewProfile.jspa?name=hilbert):_

The dark monitor keeps track of dark rates and stats on a per instrument, aperture, and readpattern basis. This information is all recorded correctly in the database tables.

However, when creating the dark monitor plots, the functions in monitor_dark_bokeh.py only pay attention to instrument and aperture. This means that statistics for different readpatterns are potentially being mixed, which will lead to confusion. This is only an issue for MIRI and NIRSpec. The other instruments collect darks with only a single readout pattern. But MIRI has darks using FASTR1 and SLOWR1 (as seen in PIDs 1517 and 1519. It looks like the other FAST* and SLOW* readout patterns are not used), while NIRSpec uses NRSRAPID and NRSIRS2RAPID. For these cases, the dark monitor should create separate plots for each readout pattern.

stscijgbot-jwql, 2023-12-20:
_Comment by [Bryan Hilbert](https://jira.stsci.edu/secure/ViewProfile.jspa?name=hilbert) on [JIRA](https://jira.stsci.edu/browse/JWQL-157?focusedCommentId=677072#comment-677072):_

Also, for NIRCam there is the option of taking full frame darks using 1 or 4 amplifiers. Right now JWQL is ignoring this option as well. I don't think NIRCam has taken any 1-amp full frame darks as of now. But it's something to think about.

bhilbert4, 2023-12-29:
I thought about including this update in #1252, but I think that's too much scope creep. I'll make this update in a new PR. I checked, and all the necessary data is being saved in the database tables (or will be, as soon as #1252 is merged). At that point, we just need to update DarkMonitorPlots in monitor_dark_bokeh.py. Need to figure out how to show all of this. It means a separate histogram for each aperture/readpattern combination, as well as a separate mean dark image for each. For the trending plot, we may be able to put all readpatterns together on one plot for each aperture.