spacetelescope/jwql #1457: Update dashboard to look at new anomaly db tables

stscijgbot-jwql, 2024-01-29:
_Issue [JWQL-168](https://jira.stsci.edu/browse/JWQL-168) was created on JIRA by [Bryan Hilbert](https://jira.stsci.edu/secure/ViewProfile.jspa?name=hilbert):_

The dashboard produces a plot showing the number of each type of anomalies in the database for each instrument. Since we switched the anomaly database into the Django models (djangomodeldb), the dashboard is now looking at the wrong location for the anomaly info. 

We should point the dashboard to the correct location. The jwqldb page also looks in the wrong place when asked to show an <instrument>_anomaly table, so we should make the same update there.

stscijgbot-jwql, 2024-01-29:
_Comment by [Bryan Hilbert](https://jira.stsci.edu/secure/ViewProfile.jspa?name=hilbert) on [JIRA](https://jira.stsci.edu/browse/JWQL-168?focusedCommentId=681460#comment-681460):_

Look here for an example of querying the Django models: <https://github.com/spacetelescope/jwql/blob/develop/jwql/website/apps/jwql/data_containers.py#L1492>

BradleySappington, 2024-02-19:
PR1491 will fix this issue WRT the Dashboard.  The JWQLDB page will remain with a bad link for the <instrument>_anomaly tables until we finally delete these tables from JWQLDB