spacetelescope/jwql #1559: Allow app to handle missing MAST files

stscijgbot-jwql, 2024-05-02:
_Issue [JWQL-194](https://jira.stsci.edu/browse/JWQL-194) was created on JIRA by [Bryan Hilbert](https://jira.stsci.edu/secure/ViewProfile.jspa?name=hilbert):_

When you try to load a page containing files that are not yet in MAST, JWQL crashes. It would be nice if the app could gracefully handle this situation, and perhaps create a page with the dummy thumbnail and N/A for the data that is displayed. 

 

Example (as of 4/25/2024):

views.view_exposure crashes when trying to load the page with jw04265002001_07101_00001 with the error:

"jw04265002001_07101_00001_nrs2_rate.fits is not in the predicted location" 

This failure is from the "filesystem_path" function in utils/utils.py, which is being called from get_additional_exposure_info() in views.py

Looking into the MAST directories, I see that the cal, s2d, and x1d files for nrs2 are present, but the uncal, rate, rateints files are missing. (Weird...) For nrs1, the uncal, rate, and rateints files are present, but everything else is missing.

stscijgbot-jwql, 2024-05-02:
_Comment by [Bradley Sappington](https://jira.stsci.edu/secure/ViewProfile.jspa?name=bsappington) on [JIRA](https://jira.stsci.edu/browse/JWQL-194?focusedCommentId=696874#comment-696874):_

handled with #1549

stscijgbot-jwql, 2024-05-02:
_Comment by [Bradley Sappington](https://jira.stsci.edu/secure/ViewProfile.jspa?name=bsappington) on [JIRA](https://jira.stsci.edu/browse/JWQL-194?focusedCommentId=696877#comment-696877):_

[Bryan Hilbert](https://jira.stsci.edu/secure/ViewProfile.jspa?name=hilbert) - This can be tested when we prep for our next release

BradleySappington, 2024-05-06:
fixed with #1549

closed 2024-05-06 by BradleySappington