spacetelescope/jwql #1612: MSATA and WATA monitors are not returning plots

stscijgbot-jwql, 2024-07-08:
_Issue [JWQL-246](https://jira.stsci.edu/browse/JWQL-246) was created on JIRA by [Bryan Hilbert](https://jira.stsci.edu/secure/ViewProfile.jspa?name=hilbert):_

Currently the pages both return "None" rather than showing plots.
<https://jwql.stsci.edu/nirspec/msata_monitor/>
<https://jwql.stsci.edu/nirspec/wata_monitor/>

stscijgbot-jwql, 2024-07-08:
_Comment by [Bryan Hilbert](https://jira.stsci.edu/secure/ViewProfile.jspa?name=hilbert) on [JIRA](https://jira.stsci.edu/browse/JWQL-246?focusedCommentId=707431#comment-707431):_

bokeh-3.3.0.min.js:230 Uncaught (in promise) 
Error: Error rendering Bokeh model: could not find #aa883134-aa51-4cba-a6b9-e06cc0dedfe8 HTML tag
    at s (bokeh-3.3.0.min.js:230:132)
    at o._resolve_root_elements (bokeh-3.3.0.min.js:230:682)
    at k (bokeh-3.3.0.min.js:163:516)
    at t.embed_items (bokeh-3.3.0.min.js:163:1376)
s	@	bokeh-3.3.0.min.js:230
o._resolve_root_elements	@	bokeh-3.3.0.min.js:230
k	@	bokeh-3.3.0.min.js:163
t.embed_items	@	bokeh-3.3.0.min.js:163

mfixstsci, 2024-07-13:
With updates to `v1.2.10` bokeh versions were updated to `bokeh=3.5.0`. We also incorporated the changes to upgrade the NRS WATA & MSATA Monitors to use the Django models method to manage our database table. The error from above is the result from the previous monitor having mismatching bokeh versions in the bokeh files.

Currently, there is a PR that should/hopefully will fix all of our bokeh plotting/NRS TA monitor woes. #1620

stscijgbot-jwql, 2024-08-29:
_Comment by [Bryan Hilbert](https://jira.stsci.edu/secure/ViewProfile.jspa?name=hilbert) on [JIRA](https://jira.stsci.edu/browse/JWQL-246?focusedCommentId=716143#comment-716143):_

This has been fixed.

closed 2024-08-29 by stscijgbot-jwql